ArduCAM/ArduCAM_ESP8266_UNO #1: CRC doesn't match. File is corrupted.

rei-vilo, 2016-06-13:
### Basic Infos

Updating ArduCAM_ESP8266_UNO from Arduino 1.6.9 Boards Manager

#### Hardware

ArduCAM ESP8266 UNO board

### Description

Error message

>
CRC doesn't match. File is corrupted.
>

### Settings in IDE

Arduino IDE 1.6.9
Mac OS X 10.11.5

### Sketch

None

### Debug Messages

None

[image]

ArduCAM, 2016-06-13:
@Rei Vilo, it is the know issue.
You have to remove the packages already downloaded in the Arduino15 folder, which is located in C:\Users\XXX\AppData\Local\Arduino15 windows system.
Not sure where it is in Mac.

rei-vilo, 2016-06-13:
Thank you for the tip. 

I deleted `~/Library/Arduino15/staging/packages/ArduCAM_ESP8266_UNO.zip` and installation went fine.

Surprisignly, this issue does"t arise for other platforms on Arduino 1.6.9.

closed 2016-06-13 by rei-vilo