ArduCAM/ArduCAM_ESP8266_UNO #3: ArduCAM ESP8266 UNO  v2 - error: espcomm_open failed

mhaleywgtn, 2016-10-02:
Hi
Just got an ArduCAM ESP8266 UNO v2 board and a 2MP camera.  When I just a basic test of using the ArduCAM_ESP8266 OV2640 capture example and tried to load it onto the board  I got the following errors

WARNING: library ESP8266WiFi claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [ArduCAM_ESP8266_UNO] architecture(s).
WARNING: library ESP8266WebServer claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [ArduCAM_ESP8266_UNO] architecture(s).
WARNING: library Wire claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [ArduCAM_ESP8266_UNO] architecture(s).
WARNING: library SPI claims to run on [esp8266] architecture(s) and may be incompatible with your current board which runs on [ArduCAM_ESP8266_UNO] architecture(s).
 

I have not made any changes from the standard settings and so assumed that it would work for the basic functionality

ArduCAM, 2016-10-03:
@Matt, please try the example and ignore the warning first.

mhaleywgtn, 2017-05-15:
working

closed 2017-05-15 by mhaleywgtn