ArduCAM/ArduCAM_ESP8266_UNO #5: When I try to update my Arducam 8266 board in Arduino using board manager, I get an error that the CRC doesn't match and file is corrupt.

NorCalAl, 2016-11-13:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Hardware:			?ESP-12?
Core Version:      	?2.1.0-rc2?

### Description

Problem description

### Settings in IDE

Module:  			?Generic ESP8266 Module?
Flash Size: 		?4MB/1MB?
CPU Frequency:		?80Mhz?
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

ArduCAM, 2016-11-14:
@AI, I don't understand you information below. For CRC error, you'd better to delete all the board package download from board manager to re-download the new one.

----------------------------------------------------------------
Please fill the info fields, it helps to get you faster support ;)
if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md
for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md
----------------------------- Remove above -----------------------------
Basic Infos
Hardware
Hardware: ?ESP-12?
Core Version: ?2.1.0-rc2?
Description
Problem description
Settings in IDE
Module: ?Generic ESP8266 Module?
Flash Size: ?4MB/1MB?
CPU Frequency: ?80Mhz?
Flash Mode: ?qio?
Flash Frequency: ?40Mhz?
Upload Using: ?OTA / SERIAL?
Reset Method: ?ck / nodemcu?
Sketch
#include <Arduino.h>
void setup() {
}
void loop() {
}
Debug Messages
messages here

NorCalAl, 2016-11-14:
Hi Lee,
I noticed my arduino IDE didn't list any of the Arducam boards anymore, so I checked the Boards Manager. It was there but not installed. 
I checked the address for the Boards in Preferences and it was correct, so I hit Install in Boards Manager. That's when I got the file corrupt message. I removed the address from preferences and restarted. Added it back in and restarted. Same thing.
Any hints?

Thanks,
Al

NorCalAl, 2016-11-14:
Hey Lee -
As a follow up, I deleted all board definitions and the board address and restarted. Readded board address to prefs. Went to Board Manager and : 
"CRC doesn't match. File is corrupted.
java.lang.RuntimeException: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:176)
	at java.lang.Thread.run(Thread.java:745)
Caused by: java.lang.Exception: CRC doesn't match. File is corrupted.
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:77)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:102)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$1(ContributionManagerUI.java:173)
	... 1 more”

That’s the output from the Arduino IDE. 

I’m not positive, but it doesn’t look like an error on my end. 
Thanks so much!
Al

NorCalAl, 2016-11-14:
OK, so I missed the zip file - deleting that, restarted and readded address. Downloaded new zip and installed just fine.

closed 2016-11-14 by NorCalAl