ArduCAM/ArduCAM_ESP8266_UNO #6: AdruCAM Mini unable to capture and upload to server

rachelgzh, 2016-11-14:
Hi Lee,

I have the ArduCAM Mini 2MP V2 board and tried your example ArduCAM_ESP8266_OV2640_Capture.
In the memory.h , i select step1: #define OV2640_MINI_2MP and step2: #define OV2640_CAM

I am able to complie and run but unable to capture and upload to server. I had change the ssid and password, this is what i get when i monitor it through serial monitor i get this

"ArduCAM Start!
OV2640 detected.

Share AP: arducam_esp8266
The password is:

192.168.4.1
Server started"

Please advise as i am using this cam for my final year project.
Thank you.

Regards,
Rachel

ArduCAM, 2016-11-15:
@rachelqzh, The example is AP mode by default, you have to use  your laptop or mobile phone to connect to the ESP8266 board and then send capture command from html page.

rachyline, 2016-11-15:
@ArduCAM Its now connecting to my wifi and i have change the address from html page. But i still cant get image when i launch the html page

theopensourcerer, 2017-05-02:
@rachyline Did you ever find out what was causing the problem?

patsadow2, 2017-10-23:
@ArduCAM How can i "The example is AP mode by default, you have to use your laptop or mobile phone to connect to the ESP8266 board and then send capture command from html page." is there any tutorial for green?

UCTRONICS, 2017-10-24:
@patsadow2 
You can refer to http://www.arducam.com/knowledge-base/esp8266-uno-tutorial-2/