ArduCAM/ArduCAM_ESP8266_UNO #7: error;

paekitt, 2016-12-15:
Hello, 

Arduino: 1.6.13 (Windows 10), Board: "nodeMCU 1.0(ESP-12E [Modue])

Build options changed, rebuilding all
C:\Program Files (x86)\Arduino\libraries\ArduCAMesp8266\examples\ESP8266\ArduCAM_Mini_OV2640_websocket_server\ArduCAM_Mini_OV2640_websocket_server.ino: In function 'void setup()':

ArduCAM_Mini_OV2640_websocket_server:59: error: 'class ArduCAM' has no member named 'begin'
   myCAM.begin();

ArduCAM_Mini_OV2640_websocket_server:71: error: 'OV2640_CHIPID_HIGH' was not declared in this scope
   myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);

ArduCAM_Mini_OV2640_websocket_server:72: error: 'OV2640_CHIPID_LOW' was not declared in this scope
   myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);

C:\Program Files (x86)\Arduino\libraries\ArduCAMesp8266\examples\ESP8266\ArduCAM_Mini_OV2640_websocket_server\ArduCAM_Mini_OV2640_websocket_server.ino: In function 'void read_fifo_burst(ArduCAM)':

ArduCAM_Mini_OV2640_websocket_server:184: error: 'class ArduCAM' has no member named 'startSend'
   myCAM.startSend();

ArduCAM_Mini_OV2640_websocket_server:186: error: 'class ArduCAM' has no member named 'sendData'
   myCAM.sendData(0x00);

ArduCAM_Mini_OV2640_websocket_server:191: error: 'class ArduCAM' has no member named 'sendData'
     temp = myCAM.sendData(0x00);

ArduCAM_Mini_OV2640_websocket_server:221: error: 'class ArduCAM' has no member named 'endSend'
   myCAM.endSend();

dprophet, 2016-12-15:
Did you follow the instructions on

http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

You need the purchase the ArduCAM esp8266 board and camera to use that example code. ArduCAM esp8266 is a very customized board.

ArduCAM, 2016-12-15:
@Paekitt, from you compilation error message, the ArduCAM library is not well configured and placed.
Like Erik said you follow the instructions and user guide find from the link below:
http://www.arducam.com/arducam-esp8266-uno-board-arduino-camera/

paekitt, 2016-12-16:
And, I use board Arduino UNO and code ArduCAM_Mini_Video_Streaming don't error. [https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_Video_Streaming/ArduCAM_Mini_Video_Streaming.ino](url) 

______Port Opened.
ACK CMD ArduCAM Start!
ACK CMD Can't find OV5642 module!
ACK CMD switch to OV5642_320x240
ACK CMD CAM start video streaming.______

But no picture preview on Application.
[https://github.com/ArduCAM/Arduino/tree/master/ArduCAM/examples/host_app](url)

I do follow the instructions on
http://www.arducam.com/arducam-mini-released/

dprophet, 2016-12-17:
Lets backup. How about you tell us what you purchased before we tell you how to fix it.

@paekitt I recommend you make a better support post. Look at the post I made asking for assistance. I give full pictures, wiring setups, hardware models, etc

https://github.com/ArduCAM/Arduino/issues/109

DISCLAIMER: I dont know ArduCAM nor do I work for them.
If you are a novice and dont understand compilers, boards, wiring, etc I recommend you get

https://www.amazon.com/Arducam-ESP8266-OV2640-Compatible-Arduino/dp/B01ESRYDYE/

Its the easiest to get started with.

When you get 1 thing working its easier to get more complicated projects working.

> ACK CMD Can't find OV5642 module!

This usually means you have a wiring problem. Your camera isnt wired up correctly.

ArduCAM, 2016-12-20:
@paekitt, from your error message "ACK CMD Can't find OV5642 module!", there must be something wrong with the camera connection.
You have to fix this issue before normal operation.

paekitt, 2016-12-21:
@ArduCAM , Now 

Port Opened.
ACK CMD ArduCAM Start!
ACK CMD OV2640 detected.
ACK CMD switch to OV2640_320x240
ACK CMD switch to OV2640_640x480
ACK CMD CAM start single shoot.
ACK CMD CAM Capture Done!
4

but don't show picture on host application of the ArduCAM_v1.0.exe .