ArduCAM/ArduCAM_ESP8266_UNO #10: Soft WDT reset

Lucho1976, 2017-03-09:
#### Hardware
ArduCAM ESP8266 UNO V2
Arducam Mini-@MP

### Description

Hi, i Have the above hardware, i set memorysaver.h enabling  #define OV2640_MINI_2MP and load Arducam_ESP8266_V2_Capture2SD and modify the set_JPEG_Size To myCAM.OV2640_set_JPEG_size(OV2640_1600x1200);

I got some jpg on he SD, but in a momment i get every time Soft WDT reset Message as below.

Do you know what it can be. Coniguration is ok ?

Regards

### Debug Messages

Soft WDT reset

ctx: cont 
sp: 3ffefa10 end: 3ffefe00 offset: 01b0

>>>stack>>>
3ffefbc0:  000000ff 00003f7e 00000258 40203707  
3ffefbd0:  3ffeead0 3ffeea50 00000258 402037e9  
3ffefbe0:  3ffeead0 00000000 3ffeea50 40203c10  
3ffefbf0:  3ffeeacc 3ffeead0 00000001 402045d5  
3ffefc00:  3ffee9d8 00000100 3fff0894 3fff0894  
3ffefc10:  00000100 00000100 3ffefc80 402044cd  
3ffefc20:  0fffffff 00000000 00000000 3ffeea60  
3ffefc30:  00000000 00000100 00000001 00000000  
3ffefc40:  000212a2 00000000 3ffeea00 00004000  
3ffefc50:  00000002 3ffeea60 00000000 000000c0  
3ffefc60:  3ffee9d8 00000000 3ffefd80 402032f2  
3ffefc70:  3ffee9d8 3ffeedb0 3ffee9e0 4020247c  
3ffefc80:  523c1752 e842a86e 281dbd37 86c4eae5  
3ffefc90:  65ada4ee 993400a2 5a420b35 638a9b12  
3ffefca0:  e63762e9 5cb54f98 21a42577 c62da7a5  
3ffefcb0:  523437ec 1426687a 714f5a1f 56d25cd8  
3ffefcc0:  f461c1ae c4b8b1a6 cd4eeddf 8415b436  
3ffefcd0:  b54a0fa7 52b08576 4163b754 ace6523a  
3ffefce0:  5a5adc95 2e24eecd ab4b6b29 4151940f  
3ffefcf0:  29a753eb fd528c22 576e2d2b 4b712ec4  
3ffefd00:  4d59ad9a 34709a09 5ac825f7 4c28c670  
3ffefd10:  50c55300 4a713834 c2bd520d 556a8a24  
3ffefd20:  4a90faa2 d49a3e0d 843e754c 5aea80fb  
3ffefd30:  96a2b0a0 343b0590 aa4ee100 4571e001  
3ffefd40:  408a3a00 0998062d a5044c4b 290a17a4  
3ffefd50:  48959400 31524941 02ec6929 dc28501a  
3ffefd60:  c0a69202 42a86214 6931d0d2 290a2045  
3ffefd70:  98e25280 02a43605 3625608a ac210598  
3ffefd80:  3ffe8468 00000000 000003e8 86941403  
3ffefd90:  706a2e32 00000067 00000000 00000000  
3ffefda0:  3fff0894 706a2e32 3ffe0067 3ffeecf4  
3ffefdb0:  00004f03 00000001 000000ac 3ffeede0  
3ffefdc0:  402015c2 00001388 00001388 3ffeedd4  
3ffefdd0:  3fffdad0 00000000 3ffeedcc 402024f7  
3ffefde0:  3fffdad0 00000000 3ffeedcc 40204f2c  
3ffefdf0:  feefeffe feefeffe 3ffeede0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

messages here
```

supprot, 2017-03-10:
@Lucho1976 
Hi,
Please try to add some yield()
 when saving the image to your SD card.

Let us know if you need more help.
Regards,
ArduCAM support team.

Lucho1976, 2017-03-12:
Hi, i'll try. I tried with 1280x and it works fine and is fine to me, but i will try however with this.

Thanks

Lucho1976, 2017-03-12:
Like this?.

void loop(){
  delay(15000);
  myCAMSaveToSDFile();
  yield();
  yield();
  yield();
}

I have the same problem, only with 1600x1200

supprot, 2017-03-13:
@Lucho1976 
Hi,

The ESP8266 has it's own operation system. You can not delay too long time to delay or to deal with one thing.You should add yield() in the  process. For example, When writting datas to SD card, you can add yield() after each outFile.write().

Let us know if you need more help.
Regards,
ArduCAM support team.