ArduCAM/ArduCAM_ESP8266_UNO #12: Compile error on WinXP

krowden, 2017-03-28:
I have a basic ArduCAm ESP8266 Uno board (no camera).  I have sketches running on Win10 in the IDE v1.8.1. It's good - I'm using the SD card for logging, and WiFi with the MySQL library for direct database access across the web.

I have a small WinXP laptop that I'd like to use on-site at remote locations - mainly for tweaking and setup purposes.  The problem is that in the same IDE (1.8.1) on WinXP the compiler fails with a missing file (see below).  I have to admit the library setup process is not particularly clear but I worked through it on Win10 and all was fine. 

On WinXP:
- I used the info in "doc/installing.md" (from here on GitHub). 
- I used the JSON definition for the board - it all works fine. 
- I ran through the Python bit and it all completed ok (I used Python 2.7).   
But when I compile a simple blink sketch I get the "no such file or directory" msg below. 
I tried copying over the "arducam_esp8266_uno" folder structure from the Win`10 "packages" folder (because it works) - but I get the same error! Weird.  Any ideas?  Thanks.

(GitHub is being a true "git" and is messing with the layout below - sorry about that, I don't know how to prevent the unwanted formatting.)

In file included from c:\documents and settings\user1\local settings\application data\arduino15\packages\arducam_esp8266_uno\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\memory:62:0,

                 from C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2\cores\esp8266\Esp.cpp:24:

c:\documents and settings\user1\local settings\application data\arduino15\packages\arducam_esp8266_uno\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>
                            ^
compilation terminated.

exit status 1
Error compiling for board ArduCAM ESP8266 UNO.

supprot, 2017-03-28:
@krowden 
Hi,

Please try delet the Arduino15 folder and restart your Arduino IDE.
Then put  http://www.arducam.com/downloads/ESP8266_UNO/package_ArduCAM_index.json
to Preferences->Additional Boards Manager URLs.
Then download our ArduCAM ESP8266 UNO biard again.

Let us know if you need more help.
Regards,
ArduCAM support team

krowden, 2017-03-28:
I did as you instructed - but got the same error.
The paths are quite long. Is there a way to put the files in another location that has a shorter path?
Here is the verbose output:

Arduino: 1.8.1 (Windows XP), Board: "ArduCAM ESP8266 UNO, 80 MHz, 9600, 4M (3M SPIFFS)"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages -hardware C:\krowden\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\krowden\Arduino\libraries -fqbn=ArduCAM_ESP8266_UNO:ArduCAM_ESP8266_UNO:ArduCAM_ESP8266_UNO:CpuFrequency=80,UploadSpeed=9600,FlashSize=4M3M -ide-version=10801 -build-path C:\DOCUME~1\user1\LOCALS~1\Temp\arduino_build_898000 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\esptool\0.4.8 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\mkspiffs\0.1.2 -verbose C:\krowden\Arduino\Blink_and_Print\Blink_and_Print.ino
C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages -hardware C:\krowden\Arduino\hardware -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\krowden\Arduino\libraries -fqbn=ArduCAM_ESP8266_UNO:ArduCAM_ESP8266_UNO:ArduCAM_ESP8266_UNO:CpuFrequency=80,UploadSpeed=9600,FlashSize=4M3M -ide-version=10801 -build-path C:\DOCUME~1\user1\LOCALS~1\Temp\arduino_build_898000 -warnings=none -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.esptool.path=C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\esptool\0.4.8 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2 -prefs=runtime.tools.mkspiffs.path=C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\mkspiffs\0.1.2 -verbose C:\krowden\Arduino\Blink_and_Print\Blink_and_Print.ino
Using board 'ArduCAM_ESP8266_UNO' from platform in folder: C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2
Using core 'esp8266' from platform in folder: C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2
Detecting libraries used...
"C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2/tools/sdk/include" "-IC:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L    -DARDUINO=10801 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ARDUCAM_ESP8266_UNO  -DESP8266 "-IC:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2\cores\esp8266" "-IC:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2\variants\ArduCAM_ESP8266_UNO" "C:\DOCUME~1\user1\LOCALS~1\Temp\arduino_build_898000\sketch\Blink_and_Print.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2/tools/sdk/include" "-IC:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2/tools/sdk/lwip/include" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-exceptions -fno-rtti -falign-functions=4 -std=c++11  -ffunction-sections -fdata-sections -w -x c++ -E -CC -DF_CPU=80000000L    -DARDUINO=10801 -DARDUINO_ESP8266_NODEMCU -DARDUINO_ARCH_ARDUCAM_ESP8266_UNO  -DESP8266 "-IC:\Documents and Settings\ [... cut ...]
                 from C:\Documents and Settings\user1\Local Settings\Application Data\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.2\cores\esp8266\Esp.cpp:24:

c:\documents and settings\user1\local settings\application data\arduino15\packages\arducam_esp8266_uno\tools\xtensa-lx106-elf-gcc\1.20.0-26-gb404fb9-2\xtensa-lx106-elf\include\c++\4.8.2\bits\stl_algobase.h:59:28: fatal error: bits/c++config.h: No such file or directory

 #include <bits/c++config.h>

                            ^

compilation terminated.

exit status 1
Error compiling for board ArduCAM ESP8266 UNO.

supprot, 2017-03-28:
@krowden 
Hi,
Please attach some picture and your detail instructions.

Regards,
ArduCAM support team.

krowden, 2017-03-28:
I have my sketchbook location set to "C:\myfolder\Arduino" in File | Preferences.
Can I set up the ArduCAM files in a "\hardware" folder or a "\libraries" folder under that location?

krowden, 2017-04-04:
Thanks for the follow-up... it looks like the latest version of the IDE to work on WinXP is 1.6.5r5.  Apparently after that version the libraries are not correct.  I installed 1.6.5r5 and the error went away.   I have another problem with the USB 2.0 driver - so perhaps I'm beating a dead horse and should upgrade to a newer laptop!