ArduCAM/ArduCAM_ESP8266_UNO #13: Light Mode, Saturation, Contrast and Brightness Example

Lucho1976, 2017-04-05:
Hi i have an Arducam ESP8266 with OV2640 and i want to try the diferents light Mode, saturation, contrast and brigthness.

I found this document with examples like below and i saw than i must write them with  myCAM.wrSensorReg8_8() function, is that correct ?. Can i make a program wich a delate between modes and capture a file picture to see the diferences?. Ideal i would be some interface program like html wich sent parameters, is there something like that?.

Thanks....

http://www.dragonwake.com/download/camera/ov2640/OV2640%20Camera%20Module%20Software%20Application%20Notes.pdf

11.1 Light Mode
**Auto**
SCCB_salve_Address = 0x60;
write_SCCB(0xff, 0x00)
write_SCCB(0xc7, 0x00); //AWB on

**Sunny**

SCCB_salve_Address = 0x60;
write_SCCB(0xff, 0x00)
write_SCCB(0xc7, 0x40); //AWB off
write_SCCB(0xcc, 0x5e);
write_SCCB(0xcd, 0x41);
write_SCCB(0xce, 0x54);

**Cloudy**

SCCB_salve_Address = 0x60;
write_SCCB(0xff, 0x00)
write_SCCB(0xc7, 0x40); //AWB off
write_SCCB(0xcc, 0x65);
write_SCCB(0xcd, 0x41);
write_SCCB(0xce, 0x4f);

supprot, 2017-04-06:
@Lucho1976 
Hi,

Yes, if you want to change the light mode, you should set the mode by using myCAM.wrSensorReg8_8().
If you want to send the image data to the html, you can refer to
 https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_Capture

Let us know if you need more help.
Regards,
ArduCAM support team.

adjavaherian, 2017-05-14:
I'm trying to do the same thing.  In my cam setup, I do something like this. 

```cpp
  #if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
    //Check if the camera module type is OV2640
    myCAM.wrSensorReg8_8(0xff, 0x00);
    myCAM.wrSensorReg8_8(0xc7, 0x00);  //AWB on
    myCAM.rdSensorReg8_8(OV2640_CHIPID_HIGH, &vid);
    myCAM.rdSensorReg8_8(OV2640_CHIPID_LOW, &pid);
  #endif
```

But my images still look really dim and green in ambient, low-light.  Any suggestions?

supprot, 2017-05-14:
@adjavaherian 
Hi,
We have update our library to add  ArduCAM_Mini_2MP_OV2640_functions.ino demo.
Please refer to this link to learn more about the OV2640 functions.
https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/mini/ArduCAM_Mini_2MP_OV2640_functions/ArduCAM_Mini_2MP_OV2640_functions.ino

Let us know if you need more help.
Regards,
ArduCAM support team.

adjavaherian, 2017-05-16:
Thanks Lee, I'll give it a shot!

fake-name, 2017-06-02:
You updated the example to support brightness and other functions, but never updated this repository (`ArduCAM_ESP8266_UNO`) to a version of the library that has the required function calls (`OV2640_set_Brightness()`, etc...).

supprot, 2017-06-03:
@fake-name 
Hi,

Thank you for your advice, we will update it. You can just copy the ArduCAM.cpp and ArduCAM.h files to your ESP8266 UNO library and replace the old files.

Let us know if you need more help.
Regards,
ArduCAM support team.

fake-name, 2017-06-03:
Maybe respond to https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/issues/19 ?

mosazahany, 2022-03-02:
i solved this problem