ArduCAM/ArduCAM_ESP8266_UNO #14: Additional features Arducam mini 2mp

kevinvinke, 2017-04-10:
Hi,
I have the Arducam mini 2 mp with esp8266 nano dev board.
I will be using this to capture images for a telecine project.
Ideally I would like to take three images per film frame with +- exposure balance and output in raw Rgb.
Can you help with settings for this please? Or is it possible to adapt the Linux driver to make calling the sccb registers easier?
Great product btw,
Thanks Kevin

supprot, 2017-04-11:
@kevinvinke 
Hi,
At present,we don't have example about 2mp in raw format, but if you want to +- exposure balance,
you can put this functions we attached to your ArduCAM library.
void ArduCAM::set_Brightness(uint8_t Brightness)
{
	switch(Brightness)
	{
		case Brightbess4:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x40);
			wrSensorReg8_8(0x7d, 0x00);
		#endif	
		break;
		case Brightbess3:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x30);
			wrSensorReg8_8(0x7d, 0x00);
		#endif
		break;	
		case Brightbess2:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))	
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x20);
			wrSensorReg8_8(0x7d, 0x00);
		#endif	
		break;
		case Brightbess1:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))	
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x10);
			wrSensorReg8_8(0x7d, 0x00);
		#endif
		break;
		case Brightbess0:
		#if (defined (OV2640_CAM)||defined (OV2640_MINI_2MP))	
			wrSensorReg8_8(0xff, 0x00);
			wrSensorReg8_8(0x7c, 0x00);
			wrSensorReg8_8(0x7d, 0x04);
			wrSensorReg8_8(0x7c, 0x09);
			wrSensorReg8_8(0x7d, 0x00);
			wrSensorReg8_8(0x7d, 0x00);
		#endif
		break;	
	}
		
}

Let us know if you need more help.
Regards,
ArduCAM support team.

kevinvinke, 2017-04-11:
Thanks for the swift reply that helps me get a bit further.

kevinvinke, 2017-04-25:
Not really a programmer which file should your attachment be put into? Ov2640 regs?

UCTRONICS, 2017-04-26:
@kevinvinke 
Hi, 
we have update our demos and library in order to support those functions.
Please give us your email address and we will send you our new ArduCAM library.

Let us know if you need more help.
Regards,
ArduCAM support team.

closed 2017-04-26 by kevinvinke

kevinvinke, 2017-04-28:
Hi,
I did send by email.
Please send to kevinvinke5x4@yahoo.co.uk thanks.
Regard Kevin