ArduCAM/ArduCAM_ESP8266_UNO #15: Unable to use since recent updates

theopensourcerer, 2017-05-01:
I had been using the ArduCAM_ESP8266 for a while successfully. 
I bought a "Arducam 2MP V2 Mini Camera Shield with ESP8266 Nano Esp-12F Evaluation Kits" from UCTronics in October 2016

Recently I had to re-flash it because I needed to change the AP configuration but I am now unable to get it it to work. The code compiles and flashes fine and the ArduCAM_ESP8266 responds when it boots correctly, e.g.

> ArduCAM Start!
> OV2640 detected.

> Connecting to <myAP>
> .WiFi connected
> 192.168.1.200

But, when I try and use the demo html page to take a picture the image comes back in an unreadable format so the browser just shows the missing image icon.
[image]

I had some PHP code which was running on a server and taking a photo every 20 minutes before this update, using PHP_CURL and the gd library, and it worked fine. 

```
echo "About to get image from camera ..." . PHP_EOL;

        $api_request_url = $url . 'capture';
        $ch = curl_init();
        curl_setopt($ch, CURLOPT_RETURNTRANSFER, TRUE);
        $api_request_parameters = array('t' => '0.' . rand());
        $api_request_url .= '?' . http_build_query($api_request_parameters);
        curl_setopt($ch, CURLOPT_URL, $api_request_url);
        curl_setopt($ch, CURLOPT_VERBOSE, FALSE);
        curl_setopt($ch, CURLOPT_HEADER, FALSE);

        $d = new DateTime();
        // Text to be printed on image (YYYY-MM-DD H:m:s)
        $text = $d->format('Y-m-d H:i:s');

        $image_file = $image_dir.$filename;

        $fp = fopen ($image_file, 'w+');
        curl_setopt($ch, CURLOPT_FILE, $fp);
        curl_setopt($ch, CURLOPT_FOLLOWLOCATION, 1);
        curl_setopt($ch, CURLOPT_TIMEOUT, 1000);

        $api_response = curl_exec($ch);
        $api_response_info = curl_getinfo($ch);
        $api_response_content_type = curl_getinfo($ch, CURLINFO_CONTENT_TYPE);

        curl_close($ch);
        fclose($fp);

        $api_response_header = trim(substr($api_response, 0, $api_response_info['header_size']));
        $api_response_body = substr($api_response, $api_response_info['header_size']);
        // Response HTTP Status Code
        echo "Response HTTP Status Code: " . $api_response_info['http_code'] . PHP_EOL;

        // Get the image file
        $jpg_image = imagecreatefromjpeg($image_file);
```

Now if I try it I get the following error:

> About to get image from camera ...
> Response HTTP Status Code: 200
> PHP Unknown error:  imagecreatefromjpeg(): gd-jpeg: JPEG library reports unrecoverable error: Unsupported marker type 0x60 in Unknown on line 0
> PHP Warning:  imagecreatefromjpeg(): '/home/alord/timelapse-camera/images/00004758.jpg' is not a valid JPEG file in /home/alord/timelapse-camera/code/getImage.php on line 119
...
> Filesize of image: 91345

This code is unchanged and used to work. Can anyone suggest a way to resolve my problem? Thanks.

theopensourcerer, 2017-05-09:
Here is the raw data of what is supposed to be a jpg image returned by the example script:

```
GET 
http://192.168.1.200/capture [HTTP/1.1 200 OK 3383ms]
Headers
Params
Response
Call Stack
/9j/4ICQysbJxoCBgYGAgICAgID/24DDgIyIiYuJiIyLiouOjYyOkp6UkpGRkqWanJaerKauraumqqmwtsW7sLPBtKmqvNK9wcfKzc7Nr7rV29TL2sXMzcr/24DDgY2OjpKQkqOUlKPKsqqyysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysrKysr/xICfgICBhYGBgYGBgYCAgICAgICAgYKDhIWGh4iJiov/xIC1kICCgYODgoSDhYWEhICAgf2BgoOAhJGFkqGxwYaT0eGHovGUsoGRoYijwrHBldLR8KSz4vKCiYqWl5iZmqWmp6ipqrS1tre4ubrDxMXGx8jJytPU1dbX2Nna4+Tl5ufo6erz9PX29/j5+oOEhYaHiImKkpOUlZaXmJmaoqOkpaanqKmqsrO0tba3uLm6wsPExcbHyMnK0tPU1dbX2Nna4eLj5OXm5+jp6vHy8/T19vf4+fr/xICfgYCDgYGBgYGBgYGBgICAgICAgYKDhIWGh4iJiov/xIC1kYCCgYKEhIOEh4WEhICBgveAgYKDkYSFobGGksHRh+Hxk6KygYiUwpGhscGJo7PS8JXi8tGKlqS04aXxl5iZmqanqKmqtba3uLm6w8TFxsfIycrT1NXW19jZ2uPk5ebn6Onq8/T19vf4+fqCg4SFhoeIiYqSk5SVlpeYmZqio6Slpqeoqaqys7S1tre4ubrCw8TFxsfIycrS09TV1tfY2dri4+Tl5ufo6ery8/T19vf4+fr/wICRiIDwgcCDgaGAgpGBg5GB/9qAjIOBgIKRg5GAv4Dh2u7Ky6nD9o201cKq4qbPm+Ot2I6k89qhjKuOlJin4JeUx4/TpdWixsiqvpTljfPSqdKL24uZjujQ9JWPlL2rk9nHsaqQvtL61LLHg9XsodjOzIncxuGU9K7rw/rAvITFof3i1rTj5cqM59zdoqCGlNa82pCU1cGSx6/Kpaar+Y2U5LXdq5makZHVrq7hts3SuNHZjJvDs9vrss7T5si718m8yrH8o5Wy2MftmJeVn63gzcryiq7ioKK8gqvN0rLjsa+/1qit28/1IrCbr6kiu6+d6sGG9q67iZIs+4UEsvDfnb2EwOSOKP/ex7ehq5Heu+kg7y7x/t7P14ad98b3reggj88GpiyxrKTyutKr1smSydXp5O6sqqGhhNTugwzUIoqMmLTri6e0nEiHmK2TtqSei6Tqqajbh9Pz96v0qu7CCioAKK2p7sOhmNb8lbzY0d2qndXUtsraxwKw5Zy+teLY4d22v7v615X/qNzfyZuJyfamo6iu+v/CkL+x5+Cx+8zX5/H+rq7v3rKwrvrW/LXMzcqOKryVpZXc1dmrsZiX3NyK17afu5XLvc3Jtoanovy9q7bEAGz6CtL31asAgQjp0vn5pADxh92uuZXU2r27brX/qprph6n6tfOXLrvanNOHiuMCzbDHn+GmuYLI3bjs6erf7Jr6iPwAkdWapGSTswqH+Mmku5e0rdLNuuj/8PWvSLadLuHZo7kGtd6ehvP4iWisxJTXxbTB2uPZl/fx26kuwNGxo4/W/GCR7svRuPrXmfep3N/P+/mYj0qYq8hsr/K4qvvQqSTitZrX0o/f+NybvLn1lNfZv/qz2c/jktjDuevUpayZoU5KrvUM5fkqpPWYzNb5rqvan7vcstZGw9ZijcmV4Jxun8Pd2pmS+bGf4qThjsXUhJirt5Vqk7qE7gTbqdyatMRMiZHUZoaIjqKoLsEsubzS5+/eqSjr7eHV2LznINaIRK//8v3fmojZ5ee9ZuKJifK556XHsUrSxZWZsNXS+N/W+7OL+Y/I6/ik56O1h6Miis+SxIXWu7yK2fOjgd2LRIefqfizzpDDYrfjq/aVz+uzza2e5mL55rDmvIvo8u3Wo5//vYXr1MzbqcvwtQiE7pmm/ekK57T8u4vK8Nzj/bPdnP8A6s3n1+Nkx+OS4NWJq5malN6g+sMqvdK6uNrNjMeD5pmN68PSueK988lsYMWpy563nAqc3cG53ffN+rGy47eR99u1It6Y6ALa3Mcq4yIE+WK/o/FKw0y0mo/3xdsO1ZT8JpL3mbj//v0q5byr8rHFu9it8JKP3Arx1PHytdXE1ZG20gAKHNK55edM+wqSLrmt9ySz5fnLx9oo1+PW0vfZSKVMLmog8PKrRMmx2Pjr2POU28z+4dK6l5Hj3czwq9YuDM7LxN61ttPBSL/4/tXEte6h4rn/AP+pDgz2LJvQq5uR2sUEAijXANXr9fawlrrBbtnInNTNjb710+Oyp8765pzAsZXbkeb718P12+H738a35/8A19nV+IWTnuKz5S7jYpsKLvQIKPWb/tgM1izlqyStILHRha3Lkcj7ms210/7k8tyRnde9TrdKww681pzE2JEC/wCJq9BMydabnYuDZJUk6yzn5bmkJNjLn5jl8dfZxIOUv9rhsNG5+dGRz/Pi8L7drUD3JPz/icj3qJn+TqmxorEsutMk8wL8Bo0AqZajtuz909bhut74kOe0u42TTPaB1eTto7QiDOfW5ra5nkD7verWpeLL+e3xvJuz/CK3BruZvc7t3Msu+fmuw82NvLWx/dPZzY0i6afJkPhILAq428i8q4fPw9UknJrp6GTIx9rcuaom57f4h/7F67/Gty6P7uu6v5ChsZG10Za5zUry1elqQKenSsfUm7et5a0KRLbphaWS/vIqmCbz+7fTDMO19LTN9ZmwCqu72KHA8daS6sHBwSrSk8WYLpDgn5a07aT07dVA2a2X8fWnxsSP3b3W4qeByLy+jcmnlO/LKL7UwAzvj6EMKrSTvGIm9p2DDvQq0rloAsvVZt381d7b47qLgbjj5WLe1Uy73GTbr8v988UAzv+rTrvmvu3Z6L2SsueV5fLu2bXTp8q5TPrc/YW817OVjc/Wuu2vsNoq7tzJrvZojeP36cKxKOT8xQbL4t2aspvvkvj5q6abmcviq+35aEahSJ3rQs9o+Pj7/K3r7ZHbl+cKzrfExUzLlijL3Malx6qy1LCp3JwKxZP3197CsJu0ifj+7hqlk/zVtLazm7MoxfSSJsq8KhjK8vyFjcWcrf3CaOW47toM+eebTmrC282T/LuF46/0uNmakbyY+fmpl72blQoq+pryocm0o6+05/n+78rpLMdGt8+wz7L0uKmR2P3hydfU/syUsYfU0qXJp5Wq9dKPZrn5Jsjc1p7VvcXKypeL9puRj/Ao3NWF/iitl8m25Pr618Wf5rCpvQyHzI3PLmr1tpLPbvJI9pC42tH3JsbH2uEKxgjtBiSbsO3r160iwJxgDpWcBo28/JTKzKOXxLFmJJfQ4rmHnJrau5DIgeQKYvwmqRD6t8P/ktva2b6f9PsO7CbrtZbp5eSazrO44tORqiSVzmjWkijM+CQGJqH+s+WnJqG0j5rn+rPdkbXj9qrL96umKOeRSpyXCum4hb0A4wpgQPy5wM3etZ2SLuHL0vqhofaxtJebtmK1/LrGvvYuAiRguAa0rbSLZufp/OtsstyamNftSLG+n+2YmMHN5c/l9gDJi/lMuprSpCDjsZKpJPeyxpus+eW+3r3F08ursLgks+euq5+q0ufH1mTr+ZOLaOWj79X5LrbdyZCDCKung4H/2OVo9bLH/d29tYO0n+r4n8WaxrjW9Ezb26WpnanqKKzSr+L7xNSyye39nr7HLsfmz9rskukox+ME9bubpS7Z2Zjyq/bZo7Wz89/y9apiq+i2vPnyj5m+6PSVsY2c1Ejltfu4pmYg57fim+Gj+Sj4lQ7V0LSjqc/4jaeLn9qhDNL356X02s+L7uwOyf/7n+vZy9Hc6aHxpQSj52KH3pzeqdPP/LeVzNO0Rmah4sgE/+i/uSapy6jP85Ak47WF34fTSNrl0LD1qW7wmayh8ZpMCtO6q/cuRPKykPL708HPrfXYm4m2LtpE9vAqkpMu6fZKbsmRsr71sO3hnaPR+Ne9ZtZGsUIIQKfsLtPRu7BA1ee6t9RC8NXdhc+U5a+5Srmbnczpj8yrtsvDDuPa5/wi0OedZOH8qx63Bq3LErDYl9azRibR1mwm7ZHLuP2rkQb8tdXDsPmesudI6LCJvsOB1C7u3k6h+eMiPuTD9tDx0re0rafMtpNI+/sK16vIl89IhaPPuaMs6dDbtev5suREv9WtMDaRtkaeIGIsxYuy5ApIItfSo6UA+tWdSulC7QbFyerpq8fW7wqVKmzz8ekm+TxG3SD/AGTVmZWSRtK4xdqVya/F0+rXlO281CTlq8vdHAL70xxs70CR9bWTCgCt2o3bh731vtbY1rjxTOgkxbPnle8kx9jS1ZfQjfuagY+11eOcoRr1qTbX62xIoc+93dPSwyjHFADdtse/QOvPZteRXLXLgZ+Xzb2hKv61ko3NLl6x0WS75b7B71Yk8MOejZEuSv5i3Oefvkibw9e9tp/4lcE2hfkKs/n5AOeDWvOzPkTz1//xLvTLyb7Po/xusZwKw9a1m8nby7C3oYfxyfeatYvRCLn6lCyn/7S3qcNIoxC/n6H8KrfVmyJE0bVI82b7vZapy1hKWppaRATn7fT21zD/AGSR5yiRwZvzwyTNssVMwedgkcXQ//jWlJfJ9qXh/DzdKpgaOvyXGp9KJMff9e/PHkCtIPnx95zVKF4mtPT8z6HN67dW9rnV77H9ySz9PtLj1blusfc0vZzHPM3Jnb2NlPbw+dEOhfH+p7XytbPH35ki25iYsyj7v8XVThSe9ImjowaNlJsQRlCFQkIg5f0ak6VC+9dc8f5o1w6jYjqplNk+arG3ZKkiOgQqu1QOv7X1nPqVIonY8CrHlYHdBJd0zb/tt4Hx8vculXQolYMg2wrlq1JMtI0QqZ2zXMvTQpWL63J4q7/5/kCrKpqzmdLbLtgA9Knr2a3Jwdk+INXtbLT7LL7lCHCDrc+rmWI6tf3F/WyT3GRK55zVHJzjvevhvc2baL7JiST/AFrf9Wq9551mfvadgZsW+fufPpWZp1pG2puxuGj96RBuZqtGTF6U/TbPLufx4Ui5pWDH4eGtz2zy/EpEGOtW/OX5tby7IIdkwzxWTvwmsfgk85YaPNqD+befLP9OuotI47XRHMu1aDIOZp23tc341hz58+Nu8dPR+6Ucy7/NzQjN2JpMAt0uj/162+3tJJD7Cry+kdXL+f/lYN70+Z2F8c8EtZVO/Yv40Dynp/y0mH4CsnLRLky/7aXH7bXjn43ptb9uqwoivibFl4uasRj5scWVKCqVmr2y1deZ+uvYQOnhsgIkCp [... cut ...] RNCXxasKbBbnTPnHFCjr8dy4r+Wp2PGetZTnLPFA9KmtLvKWg9QWse36VJfj/avwLPWw+pe02/SS0fIgnNZsvywMqalIk63wJLjQbtvRq526Zk6X/wBo0b8I6+PLQJeRuvvW/atu81zSWpomaMfTqSxGPpaQDKfUDErd05v39ZXNCJnnFLXtnCzy1buhw2YiodEugQpGTOeetYUKHKOLCkIyxxQM4Y35mvLxuGIAAsfJmqfz75mLYNHrk6GJ/5W82PPzj0oAauW5SEaH892xIvqURLWSm+4Kv7SQ52rtrZ0ewfOpg8nyJLWtmWKc4eFIh9OT5avXvb2z0NRI4eYm+KckSpnJpfSHSjTcoZXV0/Vm9SrF3bH2SEjdaonYzfBMKpvRIv7Ro6VIrWzw9OHZ99/r2iY6jZvaJN7eifqDtdSRt2hg/wD1Kmr840Jor/tGHiK6zbGzq9azqaPd+tSVTNEwpSAG0tQAVr6W/lpOp0JMz//Z
```

theopensourcerer, 2017-05-22:
Any one got ideas or suggestions?

theopensourcerer, 2017-05-31:
Bump. Do I need to provide any further information to help someone diagnose this problem?

supprot, 2017-06-01:
@theopensourcerer 

Hi,Please attach us the content your serial monitor  print.
Regards,

theopensourcerer, 2017-06-08:
@supprot Thanks.

When I request an image using the html web page example I see this on the serial monitor of the ESP8266:

```
QL change to: 2
CAM Capturing
capture total_time used (in miliseconds):30
CAM Capture Done.
send total_time used (in miliseconds):46
CAM send Done.
```

But the returned data is corrupted somehow and can not be displayed as an image.

theopensourcerer, 2017-06-08:
If I try a bigger image size it takes longer but same issue with unreadable image data:

```
QL change to: 8
CAM Capturing
capture total_time used (in miliseconds):159
CAM Capture Done.
send total_time used (in miliseconds):12933
CAM send Done.
```

UCTRONICS, 2017-06-09:
@theopensourcerer 
Hi,
We hope this two link will help you:
http://www.arducam.com/knowledge-base/esp8266-uno-tutorial/
http://www.arducam.com/knowledge-base/arducam-typical-problemsarducam-esp8266-uno/

Let us konw if you need more help.
Regards,
ArduCAM support team.

theopensourcerer, 2017-06-09:
@UCTRONICS Thanks for the response but I fail to see how either link helps. Neither suggests any solutions to my problem:

I can compile the code. 
I can flash the ESP. 
The ESP boots fine and recognises the camera module.
I can talk to the ESP and request an image. 
The device returns data. 
Unfortunately that data is not in a recognised format. 

It used to work fine but now it does not.

BluePackStack, 2017-08-10:
@theopensourcerer 
Did you try to visualize the image with the vlc-player for example? 
I have a similar problem. Streaming works perfectly in vlc, but capturing not. Also I can change the resolution, like you did. Pretty buggy Stuff...

theopensourcerer, 2017-08-11:
@BluePackStack I haven't tried that no, thanks for the suggestion. I'll have another go with it later. 

I kind of lost interest after it stopped working and have been playing around with a Raspberry Pi Zero W and a camera on that. It's much more bulky and higher power but it works ;-)