ArduCAM/ArduCAM_ESP8266_UNO #18: wd Reset

Lucho1976, 2017-05-17:
HI Guys when i set a resolution above 320x240 i get an wd reset arduino. Dou you know why?.

I have an OV2640 2 MP and running the V2_capture2sd example.

I get this error:

SD Card detected!
OV2640 detected.
Star Capture
Capture Done.
The fifo length is :67588

Soft WDT reset

ctx: cont 
sp: 3ffef9d0 end: 3ffefe00 offset: 01b0

>>>stack>>>
3ffefb80:  000000ff 00002682 0000012c 40203707  
3ffefb90:  3ffeea4c 3ffeec90 00000040 40203186  
3ffefba0:  00000010 3ffeea4c 3ffeea50 3ffeeac8  
3ffefbb0:  3ffeead0 3ffeea50 3ffeea4c 40203b95  
3ffefbc0:  3ffeead0 00000000 3ffeea50 40203c05  
3ffefbd0:  3ffeeacc 3ffeead0 00000001 402045d5  
3ffefbe0:  3ffeead0 00000000 3ffeea50 3fff047c  
3ffefbf0:  00000000 3ffe8374 000024dd 4020462a  
3ffefc00:  3ffefc20 3ffeea60 00026b85 402046f1  
3ffefc10:  3ffee9d8 00000100 3ffefc80 40204442  
3ffefc20:  0fffffff 00000000 00000000 3ffeea60  
3ffefc30:  00000000 00000100 00000001 00000000  
3ffefc40:  0013b789 00000000 3ffeea00 0002f000  
3ffefc50:  00000002 3ffeea60 00000000 0000006e  
3ffefc60:  3ffee9d8 00000000 3ffefd80 402032f2  
3ffefc70:  3ffee9d8 3ffeedb0 3ffee9e0 4020247c  
3ffefc80:  305f881c d2bc51fb f942b317 34cc9386  
3ffefc90:  1b443eae f62183da 885c98a3 b0d4305f  
3ffefca0:  f9ed66b9 22feb456 d9584d25 33deb6b6  
3ffefcb0:  9aef7fb5 92177faf 5ee5c739 78e28daa  
3ffefcc0:  dbfbe5f4 cb9a8f32 f9fe23fc d7af79a6  
3ffefcd0:  bdb8ca8f 95f54c9a 9867d838 2fc5b3db  
3ffefce0:  43edd99a 3822f9b0 9f714748 47ed3ba5  
3ffefcf0:  ca5fdd6f e58bb995 1c971cba 9bca5f74  
3ffefd00:  01fc9de6 c9eaa859 ca34ea50 4ae5a74f  
3ffefd10:  3f704f8e 43d6d20a 69848d70 7e9cec4e  
3ffefd20:  eded7954 a1a855f9 178e387b e76ad70d  
3ffefd30:  e7674ae9 37abfca0 8d387b11 764e9013  
3ffefd40:  cc53fb9c 5cdd4fec 35b92a7d e38ed025  
3ffefd50:  ddbfcd3c 378fd2c7 0e77ad9a 3324f544  
3ffefd60:  a59fe911 ffe39421 425aad00 1e5c7204  
3ffefd70:  c769a352 42de55e5 6d95dacd 683a5d22  
3ffefd80:  3ffe8468 00000000 000003e8 feefeffe  
3ffefd90:  706a2e31 00000067 00000000 00000000  
3ffefda0:  3fff047c 706a2e31 3ffe0067 3ffeecf4  
3ffefdb0:  0000e703 00000001 00000068 3ffeede0  
3ffefdc0:  402015c2 00001388 00001388 3ffeedd4  
3ffefdd0:  3fffdad0 00000000 3ffeedcc 402024f7  
3ffefde0:  3fffdad0 00000000 3ffeedcc 40204f2c  
3ffefdf0:  feefeffe feefeffe 3ffeede0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld

supprot, 2017-05-18:
@Lucho1976 
Hi,
Please try add some 'yield() ' in the code.

Regards.

Lucho1976, 2017-05-18:
I already did it on myCAMSaveToSDFile function.
it already worked fine but i format my computer, reinstall arduino and arducam and now does not work, may be different version. Now i have 2.2.2 

Is this ok the place where i add the yield(), is this ok ?
Thanks a lot.

//Read JPEG data from FIFO
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    {
        buf[i++] = temp;  //save the last  0XD9     
       //Write the remain bytes in the buffer
        myCAM.CS_HIGH();
        outFile.write(buf, i); 
        yield() ; //*****HERE 
        yield() ; //*****HERE
      //Close the file
        outFile.close();
        Serial.println(F("Image save OK."));
        is_header = false;
        i = 0;
    }  
    if (is_header == true)
    { 
       //Write image data to buffer if not full
        if (i < 256)
        buf[i++] = temp;
        else
        {
          //Write 256 bytes image data to file
          myCAM.CS_HIGH();
          outFile.write(buf, 256);
          yield() ; //*****HERE 
          yield() ; //*****HERE  
          i = 0;
          buf[i++] = temp;
          myCAM.CS_LOW();
          myCAM.set_fifo_burst();
        }        
    }

UCTRONICS, 2017-05-18:
@Lucho1976 
Hi,
You can try add yield like the below code.
We suggest you check your hardware connection firstly.
Maybe your hardware exist some bad connections.
buf[i++] = temp; //save the last 0XD9
yield() ; //*****HERE
yield() ; //*****HERE
//Write the remain bytes in the buffer
myCAM.CS_HIGH();
outFile.write(buf, i);

//Close the file
outFile.close();
Serial.println(F("Image save OK."));
is_header = false;
i = 0;
}
if (is_header == true)
{
//Write image data to buffer if not full
if (i < 256)
buf[i++] = temp;
yield() ; //*****HERE
yield() ; //*****HERE
else
{
//Write 256 bytes image data to file
myCAM.CS_HIGH();
outFile.write(buf, 256);

Regards,
ArduCAM support team.

Lucho1976, 2017-05-18:
Hi Guys it work fine with 800x600 resolution but above that same problem. I added some more yield  but the same thing.

I have an arducam V2 and a OV2640 ( REV. B) 2MP 

I tried with memorysaver.h in this two configuracion and compile the whole project but wich one is the correct:

A)

//Step 1: select the hardware platform, only one at a time
//#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC	
#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
    #define OV2640_CAM
	//#define OV3640_CAM
	 //#define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 

B)

//Step 1: select the hardware platform, only one at a time
#define OV2640_MINI_2MP
//#define OV5642_MINI_5MP
//#define OV5642_MINI_5MP_BIT_ROTATION_FIXED
//#define OV5642_MINI_5MP_PLUS
//#define OV5640_MINI_5MP_PLUS

//#define ARDUCAM_SHIELD_REVC	
//#define ARDUCAM_SHIELD_V2

//Step 2: Select one of the camera module, only one at a time
#if (defined(ARDUCAM_SHIELD_REVC) || defined(ARDUCAM_SHIELD_V2))
	//#define OV7660_CAM
	//#define OV7725_CAM
	//#define OV7670_CAM
	//#define OV7675_CAM
   // #define OV2640_CAM
	//#define OV3640_CAM
	 //#define OV5642_CAM
	//#define OV5640_CAM 
	
	//#define MT9D111A_CAM
	//#define MT9D111B_CAM
	//#define MT9M112_CAM
	//#define MT9V111_CAM	
	//#define MT9M001_CAM	
	//#define MT9T112_CAM
	//#define MT9D112_CAM
#endif 

if you think i can try another thing or add some other yield in other section of the function i will appreciate so much.

Regards

Lucho1976, 2017-05-19:
Now is working with configure A) in OV2640_1280x1024 and it's fine to me.

Thank you guys

Lucho1976, 2017-05-20:
Again i got WDT reset, i add some serial.println and it fails just in this lines:

outFile.write(buf, 256);

i added 2, 4, 8 yield() but the same problem. i tried with A) and B) memorysaver.h above described but the same thing.

Can you help me?...

thank

supprot, 2017-05-22:
@Lucho1976 
Hi,
 yes, you can use the SDFormatter software to format 
your SD card firstly.
http://sdformatter.software.informer.com/

Let us know if you need more help.
Regards,
ArduCAM support team.