ArduCAM/ArduCAM_ESP8266_UNO #27: SD card time stamp & duplicate pictures

Josephtheelder, 2018-01-19:
### Basic Infos

#### Hardware
Hardware:	Board: ArduCAM_ESP8226_UNO 
Camera: ArduCAM_Mini_5MP_Plus_OV5642_Camera_Module
Core Version:  ESP8266 library 2.2.0 	

### Description : see bottom of this message

Timestamp is wrong and camera saves duplicate images in SD card

### Settings in IDE

Module:  			ArduCAM ESP82266 UNO
Flash Size: 		4MB/3MB
CPU Frequency:		80Mhz
Upload speed:             921600
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?
Reset Method:		?ck / nodemcu?

### Sketch file directory: https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_V2_OV5642_Capture2SD/ArduCAM_ESP8266_V2_OV5642_Capture2SD.ino

### Sketch Name: ArduCAM_ESP8266_V2_OV5642_Capture2SD.ino

### Sketch: 

// ArduCAM demo (C)2016 Lee
// web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
//
// This demo was made for Omnivision OV5642 5MP sensor.
// It will run the ArduCAM ESP8266 5MP as a real 2MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO every 5 seconds 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Resolution can be changed by myCAM.set_JPEG_size() function.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 5MP shield
// and use Arduino IDE 1.5.2 compiler or above
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can only work on OV5642_MINI_5MP or OV5642_MINI_5MP_BIT_ROTATION_FIXED
//or OV5640_MINI_5MP_PLUS or ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) || defined (OV5642_MINI_5MP_PLUS) ||(defined (ARDUCAM_SHIELD_V2) && defined (OV5642_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;
// Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;
ArduCAM myCAM(OV5642, CS);

void myCAMSaveToSDFile(){
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  static int n = 0;
  uint8_t temp, temp_last;
  File file;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println("star Capture");
 while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
 Serial.println("Capture Done!");  

 //Construct a file name
 k = k + 1;
 itoa(k, str, 10);
 strcat(str, ".jpg");
 //Open the new file
 file = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
 if(! file){
  Serial.println("open file faild");
  return;
 }
 i = 0;
 myCAM.CS_LOW();
 myCAM.set_fifo_burst();
#if !( defined (OV5642_MINI_5MP_PLUS) ||(defined (ARDUCAM_SHIELD_V2) && defined (OV5642_CAM)))
 temp=SPI.transfer(0x00);
  #if defined (OV5642_MINI_5MP)
 temp = (byte)(temp >> 1) | (temp << 7); // correction for bit rotation from readback
 #endif
 #endif
 //Read JPEG data from FIFO
 while ( (temp !=0xD9) | (temp_last !=0xFF)){
  temp_last = temp;
  temp = SPI.transfer(0x00);
   #if defined (OV5642_MINI_5MP)
  temp = (byte)(temp >> 1) | (temp << 7); // correction for bit rotation from readback
  #endif
  //Write image data to buffer if not full
  if( i < 256)
   buf[i++] = temp;
   else{
    //Write 256 bytes image data to file
    myCAM.CS_HIGH();
    file.write(buf ,256);
    i = 0;
    buf[i++] = temp;
    myCAM.CS_LOW();
    myCAM.set_fifo_burst();
   }
   delay(0);  
 }
 
 //Write the remain bytes in the buffer
 if(i > 0){
  myCAM.CS_HIGH();
  file.write(buf,i);
 }
 //Close the file
 file.close();
  Serial.println("CAM Save Done!");
}

void setup(){
  uint8_t vid, pid;
  uint8_t temp;

  Wire.begin();
  Serial.begin(115200);
  Serial.println("ArduCAM Start!");

  //set the CS as an output:
  pinMode(CS,OUTPUT);

  // initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHz
  delay(1000);
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
   
  if (temp != 0x55){
    Serial.println("SPI1 interface Error!");
    //while(1);
  }
    //Initialize SD Card
  if(!SD.begin(SD_CS)){
    Serial.println("SD Card Error");
  }
  else
  Serial.println("SD Card detected!");
   
  myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK); //disable low power
  delay(100);
//Check if the camera module type is OV5642
  myCAM.wrSensorReg16_8(0xff, 0x01);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
   if((vid != 0x56) || (pid != 0x42))
   Serial.println("Can't find OV5642 module!");
   else
   Serial.println("OV5642 detected.");
   myCAM.set_format(JPEG);
   myCAM.InitCAM();
   myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
   **myCAM.OV5642_set_JPEG_size(OV5642_320x240)**;
   delay(1000);
}

void loop(){
  myCAMSaveToSDFile();
  myCAM.set_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK);  //enable low power
  delay(3000);
   myCAM.clear_bit(ARDUCHIP_GPIO,GPIO_PWDN_MASK); //disable low power
   delay(2000);
  
  
}

### Debug Messages

Hello, before I begin I just want to let you know that the only thing i modified in this program is setting the picture resolution to 5PM by changing myCAM.OV5642_set_JPEG_size(OV5642_320x240) to myCAM.OV5642_set_JPEG_size(OV5642_2592x1944), that is it. The program works great and I have no problem getting the camera to capture images. However, the picture is time stamped into the SD card and I get this generic date which is obviously incorrect: 1/1/2000 1:00 AM

**Question: How do I get the SD card to time stamp the correct time? it is crucial for my project**

In addition, when I upload the program and the camera begins to capture images, it saves into the SD card with no problem. After 2- 5 minutes ,I pullout the SD card from the slot mounted on the ESP8266, while the camera is still on/connected to my computer and insert it in my laptop to view the images. The images a clear and other than the time stamp there is no problem. I then clear all the saved images in the SD card by formatting it. After completion, I slide the SD card back to the ESP8266 slot , hit the reset button on the board and let the camera take images. After 2-5 minutes, I repeat the process of inserting the SD card into my laptop. For some reason the images are the exact images It took the first time and It dose not have any of the newly captured images. I cleared all of the previous images so it shouldn't be there and since there is no correct time stamp, I have the camera capturing images of a timer. **After a while of formatting the SD card and capturing more images over and over again, it captures 2/10 of new images and just continues to save pictures from previous run times in the SD card. At this point none of the saved pictures are continuous and are just a mix of different times  the images were captured. I suspect that the ArduCAM_Mini_5MP_Plus_OV5642_Camera_Module has an on board memory that simply re-writes that captured image and is not clearing every time it captures a new image.** I dont have a whole lot experience in coding but I suspect the problem is in the following command lines
"  /Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();          "
Its the only place I can think of. Is it possible that I have to unplug the ESP UNO every time i want to pull out the SD card? Am I doing something wrong?  Anyways, I would simply like the have the camera and SD card to accurately time stamp the images and have the images display in sequential form .  

I sincerely appreciate your time to read my problems and look forward to your response, thank you very much.

supprot, 2018-01-19:
@Josephtheelder 
Hi,
Question: How do I get the SD card to time stamp the correct time? it is crucial for my project
A: In our demo, we can't get the right time because we don't have RTC clock.
For duplicate pictures, please replace this code to  try.
void myCAMSaveToSDFile() {
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  bool is_header = false;
  File outFile;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println(F("Star Capture"));
  while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
  Serial.println(F("Capture Done."));
  length = myCAM.read_fifo_length();
  Serial.print(F("The fifo length is :"));
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //8M
  {
    Serial.println(F("Over size."));
  }
  if (length == 0 ) //0 kb
  {
    Serial.println(F("Size is 0."));
  }
  sprintf((char*)pname,"/%05d.jpg",k);
  while(k<0xffff){
      outFile = SD.open(pname, FILE_READ);
      if(!outFile) break;  //this is a new file name
      k++;
      sprintf((char*)pname,"/%05d.jpg",k);
    }
     outFile = SD.open(pname, FILE_WRITE);  //open it by writting mode.
  i = 0;
  myCAM.CS_LOW();
  myCAM.set_fifo_burst();

  while ( length-- )
  {
    temp_last = temp;
    temp =  SPI.transfer(0x00);
    //Read JPEG data from FIFO
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    {
      buf[i++] = temp;  //save the last  0XD9
      //Write the remain bytes in the buffer
      myCAM.CS_HIGH();
      outFile.write(buf, i);
      //Close the file
      outFile.close();
      Serial.println(F("Image save OK."));
      is_header = false;
      i = 0;
    }
    if (is_header == true)
    {
      //Write image data to buffer if not full
      if (i < 256)
        buf[i++] = temp;
      else
      {
        //Write 256 bytes image data to file
        myCAM.CS_HIGH();
        outFile.write(buf, 256);
        i = 0;
        buf[i++] = temp;
        myCAM.CS_LOW();
        myCAM.set_fifo_burst();
      }
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    {
      is_header = true;
      buf[i++] = temp_last;
      buf[i++] = temp;
    }
  }
}

Josephtheelder, 2018-01-20:
**Thank you for your response , unfortunately I am getting an error that reads as follows:** 

 Arduino: 1.8.2 (Windows 10), Board: "ArduCAM ESP8266 UNO, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\Users\RR\Desktop\RR\Arducam\Programs\OV5642_V2_Capture_5MP_Sequence\OV5642_V2_Capture_5MP_Sequence.ino: In function 'void myCAMSaveToSDFile()':

OV5642_V2_Capture_5MP_Sequence:64: error: **'pname' was not declared in this scope**

 sprintf((char*)pname,"/%05d.jpg",k);

                ^

C:\Users\rr\Desktop\rr\Arducam\Programs\OV5642_V2_Capture_5MP_Sequence\OV5642_V2_Capture_5MP_Sequence.ino: At global scope:

OV5642_V2_Capture_5MP_Sequence:118: error: expected declaration before '}' token

 }

 ^

Multiple libraries were found for "SD.h"
 Used: C:\Users\rr\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.3\libraries\SD
 Not used: C:\Program Files (x86)\Arduino\libraries\SD
exit status 1
'pname' was not declared in this scope

The way I compiled the sketch is as following: 
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

#if !(defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) || defined (OV5642_MINI_5MP_PLUS) ||(defined (ARDUCAM_SHIELD_V2) && defined (OV5642_CAM)))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 16;
// Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;
ArduCAM myCAM(OV5642, CS);

void myCAMSaveToSDFile() {
char str[8];
byte buf[256];
static int i = 0;
static int k = 0;
uint8_t temp = 0, temp_last = 0;
uint32_t length = 0;
bool is_header = false;
File outFile;
//Flush the FIFO
myCAM.flush_fifo();
//Clear the capture done flag
myCAM.clear_fifo_flag();
//Start capture
myCAM.start_capture();
Serial.println(F("Star Capture"));
while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
Serial.println(F("Capture Done."));
length = myCAM.read_fifo_length();
Serial.print(F("The fifo length is :"));
Serial.println(length, DEC);
if (length >= MAX_FIFO_SIZE) //8M
{
Serial.println(F("Over size."));
}
if (length == 0 ) //0 kb
{
Serial.println(F("Size is 0."));
}
sprintf((char*)pname,"/%05d.jpg",k);
while(k<0xffff){
outFile = SD.open(pname, FILE_READ);
if(!outFile) break; //this is a new file name
k++;
sprintf((char*)pname,"/%05d.jpg",k);
}
outFile = SD.open(pname, FILE_WRITE); //open it by writting mode.
i = 0;
myCAM.CS_LOW();
myCAM.set_fifo_burst();

while ( length-- )
{
temp_last = temp;
temp = SPI.transfer(0x00);
//Read JPEG data from FIFO
if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
{
buf[i++] = temp; //save the last 0XD9
//Write the remain bytes in the buffer
myCAM.CS_HIGH();
outFile.write(buf, i);
//Close the file
outFile.close();
Serial.println(F("Image save OK."));
is_header = false;
i = 0;
}
if (is_header == true)
{
//Write image data to buffer if not full
if (i < 256)
buf[i++] = temp;
else
{
//Write 256 bytes image data to file
myCAM.CS_HIGH();
outFile.write(buf, 256);
i = 0;
buf[i++] = temp;
myCAM.CS_LOW();
myCAM.set_fifo_burst();
}
}
else if ((temp == 0xD8) & (temp_last == 0xFF))
{
is_header = true;
buf[i++] = temp_last;
buf[i++] = temp;
}
}
}

supprot, 2018-01-20:
@Josephtheelder
Hi,
Sorry for our code's error,
Please define global variable char pname[20];

Josephtheelder, 2018-01-27:
Hello, I did do that but now I unfortunately I have the following error: 

C:\Users\rr\AppData\Local\Temp\arduino_build_304818/arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x4): undefined reference to `setup'

C:\Users\rr\AppData\Local\Temp\arduino_build_304818/arduino.ar(core_esp8266_main.cpp.o):(.text._ZL12loop_wrapperv+0x8): undefined reference to `loop'

C:\Users\rr\AppData\Local\Temp\arduino_build_304818/arduino.ar(core_esp8266_main.cpp.o): In function `loop_wrapper':

C:\Users\rr\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.3\cores\esp8266/core_esp8266_main.cpp:43: undefined reference to `setup'

C:\Users\rr\AppData\Local\Arduino15\packages\ArduCAM_ESP8266_UNO\hardware\ArduCAM_ESP8266_UNO\2.2.3\cores\esp8266/core_esp8266_main.cpp:43: undefined reference to `loop'

collect2.exe: error: ld returned 1 exit status

I only changed the code this way:

// Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;
ArduCAM myCAM(OV5642, CS);
**char pname[20];**

void myCAMSaveToSDFile() {
char str[8];

Please help, thank you

supprot, 2018-01-28:
@Josephtheelder
Hi,
Please refer to this code , maybe you should change some pins definition.
// ArduCAM demo (C)2017 Lee
// Web: http://www.ArduCAM.com
// This program is a demo of how to use most of the functions
// of the library with a supported camera modules, and can run on any Arduino platform.
// This demo was made for Omnivision  2MP/5MP sensor.
// It will run the ArduCAM ESP8266 2MP/5MP as a real 2MP/5MP digital camera, provide both JPEG capture.
// The demo sketch will do the following tasks:
// 1. Set the sensor to JPEG mode.
// 2. Capture and buffer the image to FIFO 
// 3. Store the image to Micro SD/TF card with JPEG format in sequential.
// 4. Come into Deepsleep mode and keep 10 seconds.
// This program requires the ArduCAM V4.0.0 (or later) library and ArduCAM ESP8266 2MP/5MP shield
// and use Arduino IDE 1.6.8 compiler or above
#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 Nano V2 board in the Tools/Board
#endif

//This demo can work on Arducam ESP8266 Nano V2 board.
#if !(defined (OV2640_MINI_2MP)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif
// set GPIO16 as the slave select :
const int CS = 2;
//Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;

const int CAM_POWER_ON = 15;
const int sleepTimeS = 10;

char pname[20];

#if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
ArduCAM myCAM(OV2640, CS);
#elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
ArduCAM myCAM(OV5640, CS);
#elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
ArduCAM myCAM(OV5642, CS);
#endif
void myCAMSaveToSDFile() {
  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  bool is_header = false;
  File outFile;
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println(F("Star Capture"));
  while (!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
  Serial.println(F("Capture Done."));
  length = myCAM.read_fifo_length();
  Serial.print(F("The fifo length is :"));
  Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //8M
[... 111 lines cut ...]
    Serial.println(F("Can't find OV5642 module!"));
  }
  else
    Serial.println(F("OV5642 detected."));
#endif
  //Change to JPEG capture mode and initialize the OV2640 module
  myCAM.set_format(JPEG);
  myCAM.InitCAM();
#if defined (OV2640_MINI_2MP) || defined (OV2640_CAM)
  myCAM.OV2640_set_JPEG_size(OV2640_320x240);
#elif defined (OV5640_MINI_5MP_PLUS) || defined (OV5640_CAM)
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5640_set_JPEG_size(OV5640_320x240);
#elif defined (OV5642_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) ||(defined (OV5642_CAM))
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_JPEG_size(OV5642_320x240);
#endif
delay(1000);
}
void loop() {
  myCAMSaveToSDFile();
  Serial.println("Come in deepSleep mode !");
  ESP.deepSleep(sleepTimeS * 1000000);//ESP8266 sleep 10s
  Serial.println("DeepSleep finish!");
}