ArduCAM/ArduCAM_ESP8266_UNO #28: recipe.preproc.macros est un schéma manquant

sin2Lattre, 2018-02-22:
### Basic Infos

#### Hardware
Hardware:			ESP-12F
Core Version:      	2.2.3

### Description

### Settings in IDE

Module:  			ArduCAM ESP8266 UNO
Flash Size: 		4MB
CPU Frequency:	80 MHz
Flash Mode:		
Flash Frequency:	
Upload Using:		SERIAL

### Sketch

int led = 13;

// the setup routine runs once when you press reset:
void setup() {                
  // initialize the digital pin as an output.
  pinMode(led, OUTPUT);     
}

// the loop routine runs over and over again forever:
void loop() {
  digitalWrite(led, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(led, LOW);    // turn the LED off by making the voltage LOW
  delay(1000);               // wait for a second
}

### Debug Messages

Arduino : 1.8.2 (Windows 7), TD: 1.36, Carte : "ArduCAM ESP8266 UNO, 80 MHz, 921600, 4M (3M SPIFFS)"

C:\D\arduino-1.8.2\arduino-builder -dump-prefs -logger=machine -hardware C:\D\arduino-1.8.2\hardware -hardware U:\Documents\Arduino\hardware -tools C:\D\arduino-1.8.2\tools-builder -tools C:\D\arduino-1.8.2\hardware\tools\avr -built-in-libraries C:\D\arduino-1.8.2\libraries -libraries U:\Documents\Arduino\libraries -fqbn=ArduCAM:esp8266com:ArduCAM_ESP8266_UNO:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10802 -build-path C:\temp\arduino_build_392180 -warnings=none -build-cache C:\temp\arduino_cache_134736 -prefs=build.warn_data_percentage=75 -verbose C:\D\arduino-1.8.2\examples\01.Basics\Blink\Blink.ino
C:\D\arduino-1.8.2\arduino-builder -compile -logger=machine -hardware C:\D\arduino-1.8.2\hardware -hardware U:\Documents\Arduino\hardware -tools C:\D\arduino-1.8.2\tools-builder -tools C:\D\arduino-1.8.2\hardware\tools\avr -built-in-libraries C:\D\arduino-1.8.2\libraries -libraries U:\Documents\Arduino\libraries -fqbn=ArduCAM:esp8266com:ArduCAM_ESP8266_UNO:CpuFrequency=80,UploadSpeed=921600,FlashSize=4M3M -ide-version=10802 -build-path C:\temp\arduino_build_392180 -warnings=none -build-cache C:\temp\arduino_cache_134736 -prefs=build.warn_data_percentage=75 -verbose C:\D\arduino-1.8.2\examples\01.Basics\Blink\Blink.ino
Using board 'ArduCAM_ESP8266_UNO' from platform in folder: U:\Documents\Arduino\hardware\ArduCAM\esp8266com
Using core 'esp8266' from platform in folder: U:\Documents\Arduino\hardware\ArduCAM\esp8266com
Detecting libraries used...
recipe.preproc.macros est un schéma manquant

Erreur de compilation pour la carte ArduCAM ESP8266 UNO

messages here

Hi,

I just try to install the board on my computer and I obtain this message. Could you help me to solve it ?

supprot, 2018-02-22:
@sin2Lattre 

Hi,
Have you try choose NodeMCU ESP-12E module instead of ArduCAM ESP8266 UNO?

supprot, 2018-02-22:
@sin2Lattre

Hi,
Have you try choose NodeMCU ESP-12E module instead of ArduCAM ESP8266 UNO from Arduino IDE?
Please choose Tools->Boards and choose NodeMCU ESP-12E module to try.