ArduCAM/ArduCAM_ESP8266_UNO #30: Does not show up in ports

2426021684, 2018-03-14:
Please fill the info fields, it helps to get you faster support ;)

if you have a stack dump decode it:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/stack_dump.md

for better debug messages:
https://github.com/esp8266/Arduino/blob/master/doc/Troubleshooting/debugging.md

----------------------------- Remove above -----------------------------

### Basic Infos

#### Hardware
Arducam ESP8266 Uno V2

### Description

Problem description

### Settings in IDE

Board: ArduCAM ESP8266 UNO
Flash Size: 		        4MB
CPU Frequency:		80Mhz
Upload Speed:               921600

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
messages here
```

I'm on Mac OS X 10.12.6 and it does not show up in ports. Connected with a usb cable.

torjoshi, 2018-05-06:
I am seeing the same problems too. ESP8266 Arducam, I don't see the port on Mac. However if I connect my Arduino UNO, I see the port. I thought it might be a problem with my micro USB cable. However when I unplug the ArduCAM and connect a phone to the MicroUSB cable the port shows up again. Any help would be appreciated.

2426021684, 2018-05-06:
@torjoshi I had to installer a driver. I believe it was: https://github.com/adrianmihalko/ch340g-ch34g-ch34x-mac-os-x-driver but I'm not sure

torjoshi, 2018-05-06:
@2426021684 .. thank you. This did work! Appreciate your help.

closed 2018-05-19 by 2426021684