ArduCAM/ArduCAM_ESP8266_UNO #34: ArduCam ESP8266 UNO V2 Flashing issue and power light

AhmadZMQ, 2018-09-11:
### Basic Infos

#### Hardware
Hardware:	ArduCam ESP8266 UNO V2 		?ESP-12? yes
Core Version:     V2 	
### Description

Problem description - power lights get high and low at all time this lead to the uploading error.

### Settings in IDE

Module:  			ArduCam ESP8266 UNO V2
Flash Size: 		4MB
CPU Frequency:		80Mhz
Flash Mode:			?qio?
Flash Frequency:	?40Mhz?
Upload Using:		?OTA / SERIAL?SERIAL
Reset Method:		?ck / nodemcu?nodemcu

### Sketch

```cpp

#include <Arduino.h>

void setup() {

}

void loop() {

}
```

### Debug Messages

```
Arduino: 1.6.12 (Linux), Board: "ArduCAM ESP8266 UNO, 80 MHz, 921600, 4M (3M SPIFFS)"

Sketch uses 219,459 bytes (21%) of program storage space. Maximum is 1,044,464 bytes.
Global variables use 31,152 bytes (38%) of dynamic memory, leaving 50,768 bytes for local variables. Maximum is 81,920 bytes.
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
error: failed sending 0xC0
error: failed sending 8 bytes
error: failed sending 36 bytes
error: failed sending 0xC0
warning: espcomm_sync failed
error: espcomm_open failed
error: espcomm_upload_mem failed
error: espcomm_upload_mem failed

```

UCTRONICS, 2018-09-12:
@AhmadZMQ 
It seems the board's serial port exists some problem.
Can you find the serial port number in the device manager?
If not, please install the ch340 driver, you can download it from 
https://github.com/UCTRONICS/Drivers.git
If still not work, please contact our salesman to give you a replacement.

AhmadZMQ, 2018-09-12:
Thanks for your reply, Yes, port no shows in device manager...

UCTRONICS, 2018-09-12:
@AhmadZMQ 
OK, maybe the ch340 chip is bad, please contact our salesman and tell us your order number.
 We will give resend you a new one.