ArduCAM/ArduCAM_ESP8266_UNO #37: no html file in ArduCAM_ESP8266_UNO_Capture

andyp04, 2019-01-30:
After completing all the steps in the ESP8266 UNO Tutorial I go to show sketch folder in Arduino IDE and there is no HTML_2640 or any other HTML for that matter. I also went through file explorer following the path given and there is still nothing. I'm not sure if something is out of date but I would really appreciate help. Thank you

UCTRONICS, 2019-02-01:
@andyp04 
You can see it form https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_Nano_V1_Capture

UCTRONICS, 2019-02-01:
@andyp04 
@andyp04
You can see it from https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/tree/master/libraries/ArduCAM/examples/ESP8266/ArduCAM_ESP8266_Nano_V1_Capture