ArduCAM/ArduCAM_ESP8266_UNO #38: Cant take 2592x1944 pics using ov5642 5mp plus

kersee112358, 2019-02-05:
Hello! 
Iam using the ov5642 5mp plus and an esp8266 v2. I am also using the UNO_V2_capture2SD sketch and it seems that even though I change the resolution to 2592x1944, using the myCAM.OV5642_set_JPEG_size(OV5642_2592x1944); the picture is not saved in the SD card and I get the following message from the serial monitor:

SD Card detected!
OV5642 detected.
Star Capture
Capture Done.
The fifo length is :511864

Soft WDT reset

ctx: cont 
sp: 3ffefca0 end: 3ffefff0 offset: 01b0

>>>stack>>>
3ffefe50:  3ffeeb30 3ffeef0c 3ffeeb38 402025ef  
3ffefe60:  3ffeeb30 3ffeef0c 3ffeeb38 4020244c  
3ffefe70:  03c023ce d43651b5 1d079f85 58ccbcea  
3ffefe80:  621734e7 0086afab 9b6ac801 1fd84d0c  
3ffefe90:  ef24a376 d30c0407 cc46e0d2 77c07e9e  
3ffefea0:  53fa383e 92329f44 bb716a1b 49c848a5  
3ffefeb0:  cf69df1b af293d3d 6dc690d6 ce1fb0eb  
3ffefec0:  3892e3a1 804a5f7f f67ec61d a8f5f6f4  
3ffefed0:  8393efc2 1042ebb7 b63c9601 e151333d  
3ffefee0:  6723db65 c99d168e 38462814 765fa9f9  
3ffefef0:  7b1bb7e1 2fc790d3 67ba1bde 19826a9a  
3ffeff00:  064273ba 0ef94e39 9aa3ac32 6f40ee76  
3ffeff10:  5694af9a 2cc64e15 80626c38 3611484b  
3ffeff20:  2b45eaee 6333ef00 cd3dfb72 7dbe773b  
3ffeff30:  c82284ea fc8ee4b1 cd637f1a 86bb3148  
3ffeff40:  9006afec ef27ab4a cff67c01 1531b15a  
3ffeff50:  d707c9f9 e0366b8a 80a1f4a7 8c71fdd2  
3ffeff60:  85415ad3 a2f5f105 24bf45c0 88814a8f  
3ffeff70:  3ffe8460 00000000 000003e8 feefeffe  
3ffeff80:  706a2e31 00000067 00000000 00000000  
3ffeff90:  3fff072c 706a2e31 3ffe0067 3ffeee4c  
3ffeffa0:  0004d163 00000001 000000ab 3ffeefd0  
3ffeffb0:  402015c2 00001388 00001388 3ffeefc0  
3ffeffc0:  3fffdad0 00000000 3ffeefb8 4020255b  
3ffeffd0:  3fffdad0 00000000 3ffeefb8 40205220  
3ffeffe0:  feefeffe feefeffe 3ffeefd0 40100718  
<<<stack<<<

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 1264, room 16 
tail 0
chksum 0x0f
csum 0x0f
~ld
ArduCAM Start!
SD Card detected!
OV5642 detected.
 
Thank you in advance

UCTRONICS, 2019-02-06:
@kersee112358 
The higher the resolution, the longer the transmission time, which will cause the board to reset.
Please try to add some yield() when reading the image data through the SPI bus.

kersee112358, 2019-02-06:
Thank you so much for your quick response! I added the yield and it worked but it captured and saved only one picture in the 2592x1944 resolution, after that it crashed viewing some weird symbols in the serial monitor. I will try to add more yield(). However, can you recommend any solution about this cause I am working on a project that requires capturing pictures over a constant interval? Thanx again!

UCTRONICS, 2019-02-09:
@kersee112358 
You can refer to our UNO_V2_capture2SD sketch

kersee112358, 2019-02-11:
Hello! 
Even if I use the  UNO_V2_capture2SD sketch as it is, only adding the yield(), I only get one picture and then I get the following. It seems that when it is trying to save the second it crashes. 

ArduCAM Start!
SD Card detected!
OV5642 detected.
Star Capture
Capture Done.
The fifo length is :8424
Image save OK.
s$lܟ<⸮c⸮⸮gn⸮dog⸮⸮

UCTRONICS, 2019-02-11:
@kersee112358 
Please attached me your code and I will help you check it.

kersee112358, 2019-02-11:
Hello! I just figured it out thank you so much!
I removed the 
  digitalWrite(CAM_POWER_ON, LOW);//camera power off
 ESP.deepSleep(sleepTimeS * 1000000);//ESP32 sleep 10s 

and it worked. Iam sorry it was so obvious and I missed it.

closed 2019-02-11 by kersee112358