ArduCAM/ArduCAM_ESP8266_UNO #40: How can I capture raw pics ?

kersee112358, 2019-02-11:
Hello! How can I capture RAW images of 5mp ? It seems that the library for the Esp8266 uno doesnt include any function.
Thank you in advance

UCTRONICS, 2019-02-12:
@kersee112358 
The 5mp support capturing RAW images, In the Arduino UNO platform, I have added some demos.
But On the ESP8266 platform, I have not added demos. You can refer to the ArduCAM library on UNO platform. **https://github.com/ArduCAM/Arduino/blob/master/ArduCAM/ArduCAM.cpp**

kersee112358, 2019-02-13:
Hello! I checked the Uno demos, I tried to copy from the uno library files  to my ESP8266 library files folder anything referring to RAW, I compile the sketch, it works but when I try the camera it captures but never saves. Maybe I made a mistake in the registers, Iam not sure. Here is the code Iam using. 

#include <ArduCAM.h>
#include <Wire.h>
#include <SPI.h>
#include <SD.h>
#include "memorysaver.h"
#if !(defined ESP8266 )
#error Please select the ArduCAM ESP8266 UNO board in the Tools/Board
#endif

//This demo can work on OV2640_MINI_2MP/OV2640_MINI_2MP_PLUS/OV5640_MINI_5MP_PLUS/OV5642_MINI_5MP_PLUS/OV5642_MINI_5MP_PLUS/
//OV5642_MINI_5MP_BIT_ROTATION_FIXED/ ARDUCAM_SHIELD_V2 platform.
#if !(defined (OV2640_MINI_2MP) ||defined (OV2640_MINI_2MP_PLUS)||defined (OV5640_MINI_5MP_PLUS) || defined (OV5642_MINI_5MP_PLUS) \
    || defined (OV5642_MINI_5MP) || defined (OV5642_MINI_5MP_BIT_ROTATION_FIXED) \
    ||(defined (ARDUCAM_SHIELD_V2) && (defined (OV2640_CAM) || defined (OV5640_CAM) || defined (OV5642_CAM))))
#error Please select the hardware platform and camera module in the ../libraries/ArduCAM/memorysaver.h file
#endif

#define MAX_FIFO_SIZE 0x7FFFFF    //8MByte fifo frame size
#define high_quality 0                         

// set GPIO16 as the slave select :
const int CS = 16;
//Version 2,set GPIO0 as the slave select :
const int SD_CS = 0;
const int CAM_POWER_ON = 15;
const int sleepTimeS = 10;

ArduCAM myCAM(OV5642, CS);

void myCAMSaveToSDFile(){

  char str[8];
  byte buf[256];
  static int i = 0;
  static int k = 0;
  uint8_t temp = 0, temp_last = 0;
  uint32_t length = 0;
  bool is_header = false;
  File outFile;
  myCAM.wrSensorReg16_8(0x4407, 0x02);
  //Flush the FIFO
  myCAM.flush_fifo();
  //Clear the capture done flag
  myCAM.clear_fifo_flag();
  //Start capture
  myCAM.start_capture();
  Serial.println(F("Star Capture"));

 while(!myCAM.get_bit(ARDUCHIP_TRIG , CAP_DONE_MASK));
 Serial.println(F("Capture Done."));  

 length = myCAM.read_fifo_length();
 Serial.print(F("The fifo length is :"));
 Serial.println(length, DEC);
  if (length >= MAX_FIFO_SIZE) //8M
  {
    Serial.println(F("Over size."));
  }
    if (length == 0 ) //0 kb
  {
    Serial.println(F("Size is 0."));
  }
 //Construct a file name
 k = k + 1;
 itoa(k, str, 10);

 strcat(str, ".RAW");
 //Open the new file
 outFile = SD.open(str, O_WRITE | O_CREAT | O_TRUNC);
 if(! outFile){
  Serial.println(F("File open faild"));
  return;
 }
 i = 0;
 myCAM.CS_LOW();

 myCAM.set_fifo_burst();

while ( length-- )
  {
    temp_last = temp;
yield();
    temp =  SPI.transfer(0x00);
    yield();
    //Read JPEG data from FIFO
    if ( (temp == 0xD9) && (temp_last == 0xFF) ) //If find the end ,break while,
    { 
      buf[i++] = temp;  //save the last  0XD9  
    
       //Write the remain bytes in the buffer
        myCAM.CS_HIGH();
        outFile.write(buf, i);    
      //Close the file
        outFile.close();
        Serial.println(F("Image save OK."));
        is_header = false;
        i = 0;
    }  
    if (is_header == true)
    { 
  
      
       //Write image data to buffer if not full
        if (i < 256){
    
      
        buf[i++] = temp;
       
        }
        else
        {
     
          //Write 256 bytes image data to file
          myCAM.CS_HIGH();
          outFile.write(buf, 256);
          i = 0;
          buf[i++] = temp;
          myCAM.CS_LOW();
          myCAM.set_fifo_burst();
        }        
    }
    else if ((temp == 0xD8) & (temp_last == 0xFF))
    { 
    
      is_header = true;
      buf[i++] = temp_last;
      buf[i++] = temp;   
    } 
  } 
}

void setup(){
  uint8_t vid, pid;
  uint8_t temp;
  Wire.begin();
  Serial.begin(115200);
  Serial.println(F("ArduCAM Start!"));

  //set the CS as an output:
  pinMode(CS,OUTPUT);
  digitalWrite(CS, HIGH);
  pinMode(CAM_POWER_ON , OUTPUT);
  digitalWrite(CAM_POWER_ON, HIGH);

  //initialize SPI:
  SPI.begin();
  SPI.setFrequency(4000000); //4MHZ
 //Reset the CPLD
  myCAM.write_reg(0x07, 0x80);
  delay(100);
  myCAM.write_reg(0x07, 0x00);
  delay(100);
  delay(1000);
  //Check if the ArduCAM SPI bus is OK
  myCAM.write_reg(ARDUCHIP_TEST1, 0x55);
  temp = myCAM.read_reg(ARDUCHIP_TEST1);
  if (temp != 0x55){
    Serial.println(F("SPI1 interface Error!"));
    while(1);
  }
  
    //Initialize SD Card
  if(!SD.begin(SD_CS)){
    Serial.println(F("SD Card Error"));
  }
  else
  Serial.println(F("SD Card detected!"));
 
  myCAM.wrSensorReg16_8(0xff, 0x01);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_HIGH, &vid);
  myCAM.rdSensorReg16_8(OV5642_CHIPID_LOW, &pid);
   if((vid != 0x56) || (pid != 0x42)){
   Serial.println(F("Can't find OV5642 module!"));
   }
   else
   Serial.println(F("OV5642 detected."));
  myCAM.set_format(RAW);
  myCAM.InitCAM();
  myCAM.write_reg(ARDUCHIP_TIM, VSYNC_LEVEL_MASK);   //VSYNC is active HIGH
  myCAM.OV5642_set_RAW_size(OV5642_1920x1080); 
  //OV5642_set_Compress_quality(uint8_t quality);

}

void loop(){
    
  delay(5000);
  myCAMSaveToSDFile();

}

UCTRONICS, 2019-02-13:
@kersee112358 
Don't worry. When I have time, I will optimize the code to support getting the RAW format image.

kersee112358, 2019-02-13:
Hi, sorry never mind I fixed it.

kersee112358, 2019-02-13:
thank you! I thought I fixed it but it only saved 1kb..

kersee112358, 2019-02-13:
I used your sketch for raw pics for 0v5642 camera. The only thing I changed from the sketch is the CS pins and it worked.
Thank you!!

closed 2019-02-13 by kersee112358