ArduCAM/ArduCAM_ESP8266_UNO #41: Wake Up ESP8266 UNO From Deep.Sleep- HOW ?

EHUbels, 2019-04-24:
I have bought 3x Arducam 2MP Plus and 3x Arducam ESP8266 UNO's V2. I uploaded the example code based on the 2MP (don't see any example for MP2 Plus), that is working, well that is to say after uncommenting the deep.sleep at the end of the code example. To make the deep.sleep work it is needed to connect the RST with D0 (GPIO16). This resets the UNO all the time (of course). How can this work? Need the camera to go to sleep and reset due to battery usage?

Thanks

Erik

UCTRONICS, 2019-04-25:
@EHUbels 
Arducam ESP8266 does not support deep sleep. The Arducam ESP8266 Nano V2 support it.

MateuszZiolek, 2019-04-28:
Yep this is bad, GPIO16 is used by Camera CS ...