ArduCAM/ArduCAM_ESP8266_UNO #42: Error downloading - Arduino unable to download

MateuszZiolek, 2019-04-28:
Hi there !

since 2 weeks i am not able to grab ArduCam board files from GIT via Arduino:
[image]

` Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
 Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
 Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
java.lang.RuntimeException: java.lang.Exception: Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: Error downloading https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zip
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more`

For example ArduCAM_ESP32S_UNO is fine.
Hash looks good, after manual download and unpack i can see:
`Invalid library found in C:\Users\-2\Documents\Arduino\libraries\ArduCAM_ESP8266_UNO: no headers files (.h) found in C:\Users\x9-x2\Documents\Arduino\libraries\ArduCAM_ESP8266_UNO`

Any clue ?

Regards

UCTRONICS, 2019-04-29:
@MateuszZiolek 
Please delete the old version and try it again.
You can delete the C:\Users\[Your Computer]\AppData\Local\Arduino15

MateuszZiolek, 2019-04-29:
> @MateuszZiolek
> Please delete the old version and try it again.
> You can delete the C:\Users[Your Computer]\AppData\Local\Arduino15

Yep now all is fine ! many thx 🥇

UCTRONICS, 2019-04-30:
@MateuszZiolek 
You are welcome.

morichu01, 2019-07-27:
I can not download the zip file from the board manager.
It was useless to delete the old version.
The error message is as follows.

https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zipのダウンロード時にエラーが発生しました
java.lang.RuntimeException: java.lang.Exception: https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zipのダウンロード時にエラーが発生しました
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zipのダウンロード時にエラーが発生しました
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more
 https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zipのダウンロード時にエラーが発生しました
java.lang.RuntimeException: java.lang.Exception: https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zipのダウンロード時にエラーが発生しました
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:175)
	at java.lang.Thread.run(Thread.java:748)
Caused by: java.lang.Exception: https://github.com/ArduCAM/ArduCAM_ESP8266_UNO/raw/master/releases/download/2.2.4/ArduCAM_ESP8266_UNO.zipのダウンロード時にエラーが発生しました
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:139)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:73)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:55)
	at cc.arduino.contributions.packages.ContributionInstaller.install(ContributionInstaller.java:101)
	at cc.arduino.contributions.packages.ui.ContributionManagerUI.lambda$onInstallPressed$2(ContributionManagerUI.java:172)
	... 1 more
Caused by: java.io.IOException: Received invalid http status code from server: 416
	at cc.arduino.utils.network.FileDownloader.downloadFile(FileDownloader.java:205)
	at cc.arduino.utils.network.FileDownloader.download(FileDownloader.java:132)
	at cc.arduino.contributions.DownloadableContributionsDownloader.download(DownloadableContributionsDownloader.java:137)
	... 5 more

Please tell me the solution with MAC OS.

UCTRONICS, 2019-07-27:
@morichu01 
Don't worry.  Today is weekend. Tomorrow, I will test it on MAC OS and reply you as soon as possible

ArducamTeam, 2019-07-29:
@morichu01 
Hi,I test install the esp8266 board on my MacBook,which installed successfully.You can delete the \Users\[Your Computer]\Library\Arduino15 and try to install again.If the installation still fails,  please leave the error messages and I will reply to you as soon as possible.

morichu01, 2019-07-29:
I deleted the specified directory and I was able to install successfully.
Thank you for the quick answer.

ArducamTeam, 2019-07-29:
@morichu01 
You are welcome.