ArduCAM/ArduCAM_ESP8266_UNO #45: espcomm_upload_mem failed

Donovator122, 2019-07-18:
Hello, I recently purchased an Arducam ov2640 mini 2mp plus and esp8266. I was able to use the camera using an Arduino uno and am now trying to get it to work with the Arducam board. I have downloaded the board information into the Arduino program, as well as the zip library. However, whenever I try to upload, I receive error messages (picture below). What would be causing this?
Thank you!

[image]
[image]

UCTRONICS, 2019-07-18:
@Donovator122 
This due to the serial port is not work normally. Please check your serial port number.

Donovator122, 2019-07-18:
I have the serial port selected that the board is plugged into. I have tried other ports as well. Is there any other firmware I need to download?

UCTRONICS, 2019-07-18:
@Donovator122 
Except CH34x driver, you don't need to add any other firmware, Have you checked the device manager?
What's more, The serial port cannot be occupied.

Donovator122, 2019-07-18:
I am operating on Mac OS Mojave, and do not have a device manager. Is there another way for this to be checked?

UCTRONICS, 2019-07-19:
@Donovator122 
You can refer to this or you can search in google about how to install ch34x driver on MAC OS system.
https://tzapu.com/making-ch340-ch341-serial-adapters-work-under-el-capitan-os-x/